openthread/wpantund #510: Install doesn't create a wpantund.service on Ubuntu 21.10

MeGreen, 2022-04-17:
I'm trying to setup a openthread borderrouter with the nRF52840 MKD USB Dongle and an Kubuntu 21.10 machine. I'm following the install.md for Ubuntu, and would expect to have a working wpantund.service in /etc/systemd/system, according to [this instructions](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/OTBR/ ) "Verify Services", but i don't have one.

**To Reproduce**
Information to reproduce the behavior, including:
1. latest git version up to 17 Apr 2022
2. nRF52840 MKD USB Dongle @ Kubuntu 21.10 with i5-3427U
3. Fresh install of Ubuntu, following instructions from install.md, except injection of following after `./configure --sysconfdir=/etc` due to missing "make" dependency
`git clone https://github.com/openthread/borderrouter`
`cd borderrouter`
`./script/bootstrap`
4. Connected to local WiFi

**Expected behavior**
Succesfull install with functioning wpantund.service

**Additional context**
I'm completely new to networking and IOT. I have somewhat knowledge with linux as my daily driver, but please forgive me if somethings very stupid in this issue. I'm just trying to get my light bulbs somewhat smart without buying an extra device with features i don't desire.

jwhui, 2022-04-19:
@MeGreen , the setup instructions you referenced are quite old. The main branch of this repo no longer supports `wpantund`.

I would suggest following the setup guide at https://openthread.io/guides/border-router

MeGreen, 2022-04-19:
The suggested guide only references to an RCP Design. wpantund is to my knowledge needed to communicate in an NCP Design.

MeGreen, 2022-04-19:
Well, my original plan is to reuse an old laptop as the processing power for the router, because RPis are quite expensive at the moment. The easiest way hardware wise seemed to me to use an NCP design via USB.

jwhui, 2022-04-19:
> Well, my original plan is to reuse an old laptop as the processing power for the router, because RPis are quite expensive at the moment. The easiest way hardware wise seemed to me to use an NCP design via USB.

RCP is ideally suited for hosts that have significant processing and memory resources (like your laptop). NCP is better suited for hosts that are not as resource capable.

MeGreen, 2022-04-19:
Thank you for the clarification. I'm very new to this level of DIY networking, even more on the hardware side of things. Do you know by chance a ressource where i could read into how to my existing USB Dongle in an RCP design? Or is it actually just a software problem that i need to figure out?

jwhui, 2022-04-19:
> Thank you for the clarification. I'm very new to this level of DIY networking, even more on the hardware side of things. Do you know by chance a ressource where i could read into how to my existing USB Dongle in an RCP design? Or is it actually just a software problem that i need to figure out?

RCP vs. NCP is mostly a firmware configuration issue. They both use the same serial protocol (spinel). The primary difference is whether most of the OpenThread logic executes on the host or on the 802.15.4 SoC. Most of the recent documentation should be updated to support the RCP model.

MeGreen, 2022-04-19:
Thank you very, very much. This has given me quite new inspiration to look into. I will close this issue then.

closed 2022-04-19 by MeGreen