openthread/wpantund #512: HomeBrew Formulae ist deprecated; manual installation is not applicable.

B3tic0, 2022-11-11:
**Describe the bug**
When going through the wpantund Installation Guide for Mac OSX for HomeBrew, I can't install it over the 'wpantund.rb'.
Can't even build manually,  

**To Reproduce**
Information to reproduce the behavior, including:
1. Pulled the files from GitHub
2. Installed all the dependencies for Mac OSX
3. Went into etc/ folder and tried to install with HomeBrew --> FAIL
4. Went to top folder to initiate the bootstrap --> FAIL.

**Expected behavior**
I expected to install, or build a functioning 'wpantund'-tool, as described. Nothing worked. Even the "[Getting started with nRF52840 MDK USB Dongle](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/getting-started/#getting-started-with-nrf52840-mdk-usb-dongle)" is explained by a MacBook wit Mac OSX. Frustrating.

**Console/log output**

When trying to install with HomeBrew in ./etc folder with: brew install wpantund.rb
Error: wpantund: undefined method `devel' for #<Class:0x00007f830361e120>  <-- HomeBrew install error.

When trying to prepare the compile-procedure over terminal with: ./bootstrap.sh
***************************** 
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
Can't exec "aclocal": No such file or directory at /usr/local/Cellar/autoconf/2.71/share/autoconf/Autom4te/FileUtils.pm line 274.
autoreconf: error: aclocal failed with exit status: 2

 *** autoreconf failed with error code 2

**Additional context**
An Update to actual System setups would be great.

B3tic0, 2022-11-11:
Before I forget, this is my System:

macOS Monterey - 12.6.1
MacBook Pro (15-inch, 2017) - i7

jwhui, 2023-02-06:
wpantund on MAC OS X is experimental. I would recommend using Ubuntu (linux).

closed 2023-02-06 by jwhui